Apollo-Replica/DSKY #1: Analysis of LED Luminosity

Aeasala, 2024-02-22:
The characteristics and behavior of the orange/white LEDs should be observed, tuning the current-limiting resistance as needed if changes are found necessary.

It has been noted that the white LED has a ~3V Vf drop, and the orange LED a ~2V Vf drop.  Planned forward current is 20mA, to be achieved with a 5V supply.

Pablo-Ortiz-Lopez, 2024-02-22:
With initial experimentation, it has been determined that white LEDs require considerably less current to achieve a similar luminosity than an orange led with a higher forward current. 

With this in mind, and the fact that maximum led luminosity is probably something we want to let the final user to decide, I propose making a table where, given a set of possible orange led resistor values, we derive:

- Orange LED forward current
- White LED resistor value
- White LED forward current

With the constraint that both sets of leds shall have similar maximum luminosities.

Aeasala, 2024-03-07:
220o for Orange
1ko for White