Apollo-Replica/DSKY #12: Microcontroller Change

Aeasala, 2024-03-01:


Pablo-Ortiz-Lopez, 2024-03-01:
After some analysis, the initial microcontroller board chosen (Arduino UNO), is deemed to be too big to fit in the area below the keys, where it's supposed to be placed. It is because of this that I propose switching to an Arduino Micro board.

The Arduino Micro is essentially a shrunk down Arduino UNO, without sacrificing any functionality that is needed for this project. 
If there is no objection I shall update the branch #8 to use this board instead.

Aeasala, 2024-03-02:
New uC implemented as of https://github.com/Pablo-Ortiz-Lopez/DSKY/pull/8 & https://github.com/Pablo-Ortiz-Lopez/DSKY/issues/7

Aeasala, 2024-03-02:
New uC implemented as of PR https://github.com/Pablo-Ortiz-Lopez/DSKY/pull/8 & Issue https://github.com/Pablo-Ortiz-Lopez/DSKY/issues/7

closed 2024-03-02 by Aeasala